StevensRockSat-C/AIR #13: [S.5] Log time with every pressure and acceleration reading

nlundquist1, 2024-02-26:
Add a time stamp to every reading from the accelerometer or pressure sensors when the data is written to the file.

nlundquist1, 2024-03-06:
Is this to every individual value or for every group of channel readings or? How does this get formatted into the csv?

ALEEF02, 2024-03-09:
It should be thought that each line of readings happens at the same time. That time should be logged in a column in the CSV.

Please see the MPRLS implementation for getting T+ from the RTC. (I.e. `rtc.geTPlusMS()). However, this might be too slow for the needed realtime datastream. What you can do instead is get the time of the first and last read, and then interpolate it over that buffer's samples.

> Is this to every individual value or for every group of channel readings or? How does this get formatted into the csv?

ALEEF02, 2024-04-15:
We **will not** be logging the time with each acceleration. This due to:

1. We have sunk multiple weeks of full work into this feature, yet we are still seeing incontinous timings. To continue with it would be with the sunk cost fallacy.
2. We can get the time of launch from the G-switch, or other methods, and extrapolate over the entire sample, since we know the data itse is continuous (#18)

closed 2024-04-15 by ALEEF02