StevensRockSat-C/AIR #55: reqFunctional3 - BleedTank

nlundquist1, 2025-01-27:
Functional Requirement 3 (reqFunctionalBleedTank) The system shall utilize a bleed tank to clear out intermediary tubing between each sample.
• Need: Stagnant air is guaranteed to build up the intermediary tubing due to the fact that each sample pressure is decreasing subsequently thus the pressure from the previous sample will remain in the tubing when that next sample happens.
• Source: Experiments Team
• System Test Plan: How will we test that this requirement has been met?

*Might be deleted if we choose not to use a bleed tank

ALEEF02, 2025-02-13:
On hold while bleed tank solution is being worked out by mech team.

ALEEF02, 2025-03-24:
No longer on hold – Req was re-written to specify any active bleeding, i.e. static to dynamic bleed

PaniniB03, 2025-04-06:
Currently setting up a test to simulate DPP to SPP bleed